thegridelectric/gridworks-scada #164: hw1-1.electricity.works went away around 7 PM 2023-01-27?

anschweitzer, 2023-01-28:
Almond log just reports many mqtt disconnect messages (1/minute), and doesn't connect. Ear is reporting nothing has been heard for an hour. Haven't checked orange yet. mosquitto_sub doesn't fail, but just seems to sit there. MQTT Explorer connect just seems to hang.

jessicamillar, 2023-01-28:
When I cleared the dummy_ear_queue (which had something like 800,000 messages) everybody started talking again. Unfortunately I did not first go to the overview page to inspect memory, but it seems likely that the dummy_ear_queue had clogged up our memory.

queues are not meant to be long on rabbit. The dummy ear queue is just for debugging, so maybe we could leave it in with a short queue if we want to take the time to figure out how to do that. For now I am deleting that queue

jessicamillar, 2023-01-28:
Its interesting that gwd events show stopped at 3 pm but the ear didn't generate a nothing heard for the last hour until 7 pm. We should dig into that perhaps.