thegridelectric/gridworks-scada #233: Remove pendulum depdenency

anschweitzer, 2024-04-23:
Pendulum 3.0 has [issues](https://github.com/sdispater/pendulum/issues/784#issuecomment-1871714457) on 32 bit systems; our raspberry pis use Arm v7l, which is 32 bit. Pendulum < 3.0 does not support python 3.12. We don't need 3.12 yet, but in general we might have less headache if just extract pendulum dependencies.

karlknaub, 2024-04-23:
Hi,

Can I please be unsubscribed from all github notifications?

Thanks,

Karl

anschweitzer, 2024-04-23:
I think you unsubscribe yourself, but I’ll check when I’m online later this afternoon.

anschweitzer, 2024-04-24:
Hi @karlknaub, I think you have to undo what you "watch". The way I handle noisy github is to look at the bottom of the unwanted email. There's usually a "manage your subscriptions" link there which should take you to the offender.